JQIamo/SetList #77: unnecessary triggers

AceChenX, 2015-02-22:
While Step is nonzero, triggers to AO boards and Novatechs keep pulsing even if nothing is changed; that means hardware memories (NI card buffer/Novatech table) will exhaust if one row is set to ramp for any other device.

ZJ, 2015-07-21:
Are you still experiencing this?

Step being non-zero producing triggers is intended behavior.
The behavior for triggering works as follows:
 1. Every device on a given trigger channel gets a chance to declare it needs triggers
 2. If _any_ device on that channel needs a trigger, _all_ devices will receive them.  This is a reflection of the software to hardware mapping.  A shared channel is a shared BNC, so every physical device will see the triggers.

So:
 - If you're seeing that devices set up to use the same trigger channel all getting triggered together, that's the intended behavior.
 - If you're seeing triggers transferring across channels that's a bug we've not seen here.

I can't promise a quick turn around (you've probably already noticed that, though), but if this is an unresolved issue that isn't explained above, I'll take a look at your stuff.  Upload a copy of your procedure file somewhere, and leave a link here in a comment along with a detailed description of the behavior you're seeing.

ZJ, 2015-07-23:
Just to avoid confusion in this issue, the above commit (c3d1ee2ef4eb906edf8f263e2397530bb4166f52) was actually referring to issue #70, and doesn't really address this at all.

closed 2015-12-07 by ZJ