JQIamo/SetList #92: Analog Output on NI Cards

ZJ, 2016-04-22:
Trey would like working AI data from the NI cards.  The current situation, as per a discussion we just had:
  - AI acquisition in HW works out of the box
  - Clear Hardware.vi seems to read the samples in, but somewhere in 
[image]
The output values turn into an empty array
  - Nothing is hooked up to do the actual output.  The "old way" was a shared variable hooked onto AI Data, but with multiple NI Cards this might induce data race conditions.

ZJ, 2016-04-22:
Pragmatically generated shared variable with device name: <DeviceName>_<AI Data> or sim.

npisenti, 2016-04-22:
> Pragmatically generated shared variable with device name: _ or sim.

We should also see if we can move the shared variable library server ip
address to some kind of config file... I believe at the moment it's hard
coded with the strontium server IP.

ZJ, 2016-04-23:
> at the moment it's hard coded with the strontium server IP

Is that hard-coding on the publishing side or the receiving side?  I seem to recall needing to edit in the camera acquisition programs, but then again you only ever set up shared variables the one time.

dsbarker, 2016-04-23:
It's on the publishing side. The issue is that if you have TCP/IP devices
attached to the computer, Labview will often publish the variables to one
of the device IPs instead of to the other computer. Hardcoding the target
IP fixes the problem. A config file would be nice, if someone can figure
out how to set it up.

ZJ, 2016-04-23:
Ah, OK, that can be put into the preferences pane with the other one-time config type settings.  Will do.

ZJ, 2016-04-27:
Whoops, wrong direction on the title.

commit 593536c3 (2016-04-29): Add Analog Input to Shared Variable to NI Device

Wired in the automatic creation and updating of Network-Shared variable on
NI Cards.

Closes #92
closed 2016-05-09 by RbDisorderLab