JQIamo/SetList #98: Adding a "Collect Data" required override VI to Device class

ZJ, 2016-05-19:
@JQIamo/setlist-admins 

This means touching every device class, so I want to make sure it is a good idea before doing it.

Currently our programming/resetting sequence for a single cycle is something like this:
Parse Table->Request Triggers->Prune HWI->Build HWI->Load->Start->Wait for Done->Clear HW

I'd like to add a phase between the last two stages, so it would look like:
...->Wait for Done->Collect Data->Clear HW

This would allow moving the Analog Input function of the NI card to a place you'd actually think to look for it.  It also makes it possible to make sure all data is collected before CycleDone is issued.

Feedback appreciated, otherwise I'll probably just do it tomorrow.

dsbarker, 2016-05-19:
This seems like a good idea since we could also use this to ensure cameras
are finishing acquisition correctly.

ZJ, 2016-05-20:
I'm waiting to test until after DAMOP, but otherwise this is done.

commit cf2f4f78 (2016-07-05): Add "CollectData" phase to CycleX; fix error handling

CycleX now has a CollectData phase before indicating CycleDone but after
the cycle has finished to allow for acquisition of analog input (or other)
data.

Also:
  - Removed an Error/No Error case that was essentially a NOP
  - Fixed Clear Hardware loop so it always executes for all devices, even
    if some of them throw errors.
  - Labelled some SubDiagrams and took away the corresponding floating
    comment boxes

Closes #98.
closed 2016-07-05 by ZJ