JQIamo/SetList #104: Slaved digital line mismatch between SWI and hardware behavior

jqi-sr-lab, 2018-01-19:
This appears to be a rather pernicious bug, but here's the general behavior:

For slaved digital lines, if you probe the pulseblaster device in the "Start Experiment" vi, you can see for a given master channel number what the output states should be on each line of the setlist. This software version is as expected, but when you look at the output on a scope, it is missing transitions in weird places. Furthermore, the behavior is not the same from channel to channel. I wonder if there is an off-by-one error somewhere in the device programming (??) which is causing this error, but I'm at a loss.

-NCP

dsbarker, 2018-01-19:
We once saw channels being triggered before they were supposed to be. It turned out that there was cross-talk between two clock channels (due to bad BNC cables) and one of the slave devices was being triggered early as a result. This issue may be related, but it sounds like you're experiencing the opposite effect.

npisenti, 2018-01-20:
Was that on the NI cards? I feel like the pulseblaster should just output the right thing, since these are just DIO outputs from it. I don’t know enough about the API though to piece through what is actually being programmed...

Maybe we need to update it’s firmware :P

dsbarker, 2018-01-20:
Yes, so not related then. We also saw "off-by-one" errors early on when the 'wait' instruction was not implemented in the new setlist version. You might want to switch your sequence to contain only 'ramp' instructions and see what happens.

I think that instruction types that are not implemented get skipped by the pulseblaster but not by the NI cards, which can cause strange behavior.

Also, the pulseblaster has a limited number of instructions that it can contain in memory and it appears to use a FIFO buffer, so the start of your sequence can be dropped if it's too long (this is true at least on the older pulseblaster that I have at NIST). Setlist may throw an error if a sequence is too long, but labscript doesn't (although my version is very hacked to allow old pulseblasters to work at all).